jrsteensen/OpenHornet #1020: [Obsolescence]: TLVH4200 LED Red for landing gear assembly

DenzilLee, 2024-01-17:
### Discord Username

DenzilLee

### Issue Summary

Red Led TLVH4200 is an obsolete part. #23 in the PDF screenshot

### Applicable COTS Part Number

TLVH4200

### Applicable OH Part Numbers

OH4A2A1-1

### Screenshots/Images/Files

[image]

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.

jrsteensen, 2024-02-25:
Will replace with [HLMP-1321](https://www.digikey.com/en/products/detail/broadcom-limited/HLMP-1321/637558)

jrsteensen, 2024-02-25:
No change to form/fit/function. Drop in replacement.

Closed with d4323e292d2a7a41cf9475cbe7a06dd0b00ae403

commit d46fd5a2 (2024-02-25): Replaced TLVH4200 LED with HLMP-1321 LED.

No changes for form/fit/function. Drop-in replacement.
Closes #1020
closed 2024-02-25 by jrsteensen