jrsteensen/OpenHornet #1021: [Bug]: OH4A9-301 - PCB, CONTROLLER, THROTTLER Manufacturing files contain old pro-micro BOM

Arribe, 2024-01-26:
### Discord Username

Arribe

### Bug Summary

The Manufacturing folder in the ECAB and beta1 release electronics folders have the old pro-micro based BOM, and manufacturing notes say to use it for ordering.  This is causing confusion as the correct ESP32-S2 based BOM is also in the folder.

### Expected Results

Only the current files should be in the folders

### Actual Results

Old version of the BOM is present.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH4A9-301

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Arribe, 2024-01-26:
In progress...PR submitted.

Arribe, 2024-02-08:
@jrsteensen with the throttle group buy, there may be an increase in PCB orders...this issue and PR will clean-up the throttle controller's JLCPCB folder.

jrsteensen, 2024-02-13:
Closed with #1023. Thanks @Arribe

closed 2024-02-13 by jrsteensen