jrsteensen/OpenHornet #1025: [Bug]: Transposed Pins on connectors between OH4A9-125 - THROTTLE_INNER_GRIP to OH4A9-301 - THROTTLE CONTROLLER PCBs

Arribe, 2024-01-27:
### Discord Username

Arribe

### Bug Summary

The SDA / SCL lines between the throttle controller PCB and the inner grip PCB have the SDA and SCL lines.  This causes the wiring to not be a straight through connector with each pin serving the same function.  To make I2C work properly the wires need to be crossed between pins 2 and 3.

### Expected Results

Ideally there would be straight through wiring where each pin servs the same function between the PCBs.

### Actual Results

Inner grip pin 2 is SDA, where throttle controller SDA is pin 3.  Vice vera for SCL.  In the interim it is possible to make the PCBs work together by crossing the wires between pins 2 and 3 on each end of the JST connector.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH4A9-125, OH4A9-301

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-02-13:
Will likely be a beta2 mod, but we'll see. Assigning it to the beta1 project so it stays on the radar.

jrsteensen, 2024-02-25:
Updated inner throttle grip controller.

Closed with d4323e292d2a7a41cf9475cbe7a06dd0b00ae403

commit d4323e29 (2024-02-25): Swapped SCL/SDA pins on output connector.

Close #1025
closed 2024-02-25 by jrsteensen