jrsteensen/OpenHornet #1028: Missing dimension on the Wing Fold assembly drawing

mmanlove1, 2024-02-22:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
On page 6 of the OH5A2A2-1 - ASSY, WING FOLD HANDLE - v7 drawing, the dimension to locate the groove for the ball-nose spring plunger detent from the end of the shaft is missing. 

### Expected Results/How It Should Work:
Add the dimension from the end of the shaft to the location of the detent groove. 

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ X] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A2A2-1 - ASSY, WING FOLD HANDLE_-_v7.pdf

jrsteensen, 2024-02-25:
Drawing error - there is no notch machined in. Its the hidden lines showing the profile of the hole.

Redrew for clarity.

[image]

jrsteensen, 2024-02-25:
Closed with 6b44c8de6b8277d7a5f8693b546fd0659113636a

closed 2024-02-25 by jrsteensen