jrsteensen/OpenHornet #1029: [Bug]: 4A5A2 ABSIS_ALE Potential Issue with APU panel's relay board

Arribe, 2024-02-25:
### Discord Username

Arribe

### Bug Summary

Ran into trouble with the 3 electromagnets having enough holding power.  The APU will hold until the green light is on.  Then when moving the engine crank switch and turning on its 2 electro-magnets the APU holding power reduces and the switch falls to off.

### Expected Results

The APU mag should hold until the release logic is reached (i.e. 1 min past 2nd engine start and light turns off), while also having the engine crank switch held left or right until the engine's rpm >= 65%.

### Actual Results

After a few months the APU mag no longer holds, and the engine crank switch has always had challenges holding the switch.  Running 3 mags off one Relay board, and 2 mags off a single channel appears to be too much for each mag to have enough holding power.

APU holds fine when it's the only mag running.  A separate relay for just the engine crank switch should work as it will allow the L and R to be split across 2 channels so only 1 runs at a time.  There may be an option to add a relay board to the FCS panel to allow driving the engine crank mags one at a time given that there are multiple open pins on that panel.  Though then wiring won't be isolated to one physical panel.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

4A5A2 ABSIS_ALE

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2024-02-25:
The holding power of the magnet is directly related to the total surface area the magnet has to hold onto as well as the distance from the magnet.

I found for mine I had to adjust the distance so the arm was virtually right on the magnet (ie almost no distance)   do this on ehte dual switch was very difficult.  I also found there was some slippage in the mag brace as there is no anti-rotation built into it.

I don't know if this can be part of the issue or not but both the magnets and the mosfet are powerful enough to hold.
granted they do need allot of power >1amp each so with all 3 on that's 3-4amps going thru them.
the MOSFET is rated 120amp.

The last thing is the duty cycle of the magnet,   most are no designed to be on 100% of the time and have to be off for some percentage of the time.   ie a 50% duty cycle @ 10 minutes means the magnet can be on for 5 minutes, then must be off for 5 minutes before turning back on.

Arribe, 2024-02-25:
Mags could be it...Though I know my ABSIS Relay went bad...just don't know how.

I have adjusted the mags to be as close to the arm as possible (i.e. slight rub).  These mags are on for a longer duration...and with both engine crank mags coming on at the same time can lead to longer than 3.5 min when considering starting the L and R engines take time.  And if one is doing other preflight checks between R and L engine start the APU is going to be on longer than 3.5 minutes.  So if duty cycle is an issue splitting the R and L mags to be separate channels could help.  May have to search for other mag options...or find real mag switches (just not sure about costs and risks for buying used).

jrsteensen, 2024-02-25:
Another thing we may need to increase holding forces by enlarging the "target" for the mag. Our current target is rather undersized, and could be improved. The answer could be a 3D printed flat "sleeve" that we could fill with little steel ball bearings or something if we wanted to avoid machining something. (Random off the cuff thought.)

kbastronomics, 2024-02-25:
Maybe look at a different method of holding.   what about a holding pin instead of a magnet.   something like what's done with the landing gear mech.