jrsteensen/OpenHornet #1034: [Obsolescence]: Bearing Availability from McMaster

DenzilLee, 2024-03-04:
### Discord Username

DenzilLee

### Issue Summary

60355K701 BEARING, SEALED, .25 SHAFT X .625 OD X .196 THK, R4-2RS used in throttle quadrant is no longer available from Mcmaster Carr.

### Applicable COTS Part Number

60355K701

### Applicable OH Part Numbers

OH4A9-1

### Screenshots/Images/Files

[image]

### Release Version

v0.2.0

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Throttle

### Built to print?

- [ ] I built (or attempted to build) the part to the OpenHornet print without any deviations.

kbastronomics, 2024-03-04:
This was a R8-2RS bearing so the replacement is a
**60355K691**

https://www.mcmaster.com/60355K691/

jrsteensen, 2024-03-04:
Please list drawing rev and FNs in the descriptions in obsolescence reports for clarity moving forward - I will likely add it to the form when I get the chance.

So both FN 23 and 56 appear to be obsolete.
1. FN 23 (R8-2RS Bearing) Replacement is correct as [60355K691](https://www.mcmaster.com/60355K691/)
1. FN 56 (R4-2RS Bearing) Replacement is [60355K336](https://www.mcmaster.com/60355K336/)

DenzilLee, 2024-03-05:
>    2. FN 56 (R4-2RS Bearing) Replacement is 60355K336

Yes sir. Not a problem at all. Saw Mcmaster's suggested replacement,just
wanted to draw attention to it. The rpm values and axial loads were
different in the suggestion,although I realize in their current
implementation, neither of those apply, as it's never gonna see 16k rpm or
anywhere near that load in a throttle.I also saw the FN 56 as i was filling
out my Mcmaster order and I also came across the DSTL-0840-05 Solenoid no
longer being manufactured, i did find 2 which I ordered, but soon they will
be obsolete as well.