jrsteensen/OpenHornet #1036: [Bug]: Radar Altimeter Servo susceptible to burning out due to over voltage.

kbastronomics, 2024-03-06:
### Discord Username

Sandra

### Bug Summary

The Micro servo used to the control the FLAG in the Radar Altimeter is a 2.8-4.2 volt (see spec below)
driving this servo at 5V can cause it to burn out.

Specifications:
Voltage: 2.8V-4.2V 
Speed:0.07s / 60 ° 2.8V and 0.05s/60 ° 4.2V 
Torque:0.05Kg/2.5V and 0.075Kg/4.2V
Size: 16.3mm x 13.5mm x 6.3mm
Weight: 1.9g
Plug: micro JST 1.00 pitch

### Expected Results

The servo should be driven within its specified range of 2.8.4,2 volts

### Actual Results

in talking with Ben-F111 he has reported that he's blown multiple servos due to over voltage.   
We should like the voltage going to the servo.   this is both at the +V pin and the PWM pin.  

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

HK-5320 - SERVO, DIGITAL, ULTRA-MICRO, 1.7g

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-03-09:
@kbastronomics Do we have a PN for the PCB side micro JST connector?

A quick look is that there are 1mm spacing JST connectors, but micro JST is 1.25mm spacing. So need a little clarification.

Updated the circuit to the following:
[image]

jrsteensen, 2024-03-09:
Looks like a Micro JST with 1.25mm spacing.

[image]

jrsteensen, 2024-03-10:
Need to flip pinout; didn't realize pinout in datasheet was viewed from below the PCB.

PR #1037 (2024-03-10): Changed servo level from 5V to 3.3V
# Description

Revised PCB to Version 3
Added MOSFET to change levels of OFF SERVO from 5V to 3.3V. 
Changed OFF SERVO connector to 1.25mm "Micro JST" (AdamTech 125SH-V-03-TS).

Resolves #1036 

## Motivation and Context
Servo is rated for 2.8V - 4.2V. Was powered by 5V which could lead to damage.

## Type of change

Please delete options that are not relevant.

- [ ] MCAD - New Part (new Mechanical Model)
- [ ] MCAD - Revision (Revision of an MCAD model)
- [ ] ECAD - New PCB (A new PCB!)
- [X] ECAD - Revision (Revision of a PCB or the interconnect)
- [ ] Do [... cut]
commit 187908c1 (2024-03-10): Swapped pins 1 & 3 on J15

Swapped pins 1 & 3 on J15.
Closes #1036
closed 2024-03-10 by jrsteensen