jrsteensen/OpenHornet #1038: [FEATURE REQUEST]: Update 5A7A1-SNSR_PANEL Wiring

Arribe, 2024-03-11:
### Discord Username

Arribe

### Feature/Enhancement Summary

Update the Sensor Panel to either use 2 ALEs or switch to an ESP32.  Not hooking up the Radar and INS rotaries' off positions makes coding complexity logic challenging and will increase the effort it will take to maintain the OH software.  If the off positions were connected to pins then the logic would be trivial built-in DCSBios multi-position switch calls.

### Screenshots/Images/Files

[image]

### Justification

Keeping track of 2 separate rotaries to deduce if they're pointed to off, and not just in the middle of a rotation adds to the code's complexity.  This panel also needs logic for the LTD/R mag-switch.  The amount of time and effort to maintain or update the code to add new features / HID support will only be even more complex.

### Applicable Part Numbers

5A7A1-SNSR_PANEL

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Right Console (RCON)

kbastronomics, 2024-03-11:
I don't believe adding another ALE will solve this.  
But you can test this, however the code is basically bypassing DCS-BIOS and writing its own method of controlling the messages sent to DCS (see loop() in the code) .  we should stick with stock DCS-BIOS calls unless there is a specific documented reason to not do so.

The use of arrays is also complicating troubleshooting as you cannot see clearly what's happening.
   
Try using a supplied DCS Arduino Library call before we say there is an issue. which has code already for a non-shorting switch?

const byte radarSwPins[3] = { RDR_STBY, RDR_OPR, RDR_EMERG }; 
DcsBios::SwitchMultiPos radarSw("RADAR_SW", radarSwPins, 4);

const byte insSwPins[7] = { INS_CV, INS_GND, INS_NAV, INS_IFA, INS_GYRO, INS_GB, INS_TEST };
DcsBios::SwitchMultiPos insSw("INS_SW", insSwPins, 8);

if issue persist, we should open an issue with DCS-BIOS

but even so,
adding in the off pin will not solve this the switch used is a non-shorting switch, that means between switch changes nothing is connected, not that the off pin is connected.   (something be hidden due to use of arrays)

but to test connect the off pin to another I/O.   it will disable one of the switches so make sure to disable that check
example LST_ON
comment out 
DcsBios::Switch2Pos lstNflrSw("LST_NFLR_SW", LST_ON);
now use LST_ON in your code for the off position as a test.
does it solve the issue???

kbastronomics, 2024-03-11:
I think i see what you're saying now.
that the OFF position and the Non-Shorting appear as the same event

kbastronomics, 2024-03-11:
one work around but would mean not using DCS-BIOS SwitchMultiPos would be to connect off to pins INS_CV and INS_TEST
then check for both ON for the off position. otherwise if only 1 is on then you're in that position.

Arribe, 2024-03-11:
I ran a second pro micro hooked up to just the radar switch for 8 months. This allowed for both rotaries to have all pins connected. With all the positions connected the stock DCSBios calls work as you suggest, and is what I had for my personal code.

kbastronomics, 2024-03-11:
i wonder if we'll ever implement the pull function of that knob

Arribe, 2024-03-28:
Not needed, for this panel the code can overcome the PIN_NC limitations.

closed 2024-03-28 by Arribe