jrsteensen/OpenHornet #1040: [Bug]: OH4A9-1 Throttle Detent Solenoids cannot overcome spring strength.

Arribe, 2024-03-18:
### Discord Username

Arribe

### Bug Summary

The throttle detent solenoids do not have enough pulling power to overcome the spring's rating.  The detents do not pull down.

The throttle solenoids were moved up in the design to account for their short pull range.  This means they have to be at the upper end of the slots to be in position to pull a detent that is in the up position.

According to the spring rate and the solonoid positioning means that the spring is nearly fully loaded when the solenoid is off.  The solenoid would need ~19lbs of force to finish the pull.

The solenoid spec sheet shows that a 1/2" pull is ~1/2lb of force.

### Expected Results

The solenoids should be able to pull the detents down.

### Actual Results

Attempted multiple tests with 5v power to the solenoids, and code to set the solenoid pins to always be HIGH.  With the full spring, even with manual assistance the detents are very hard to push down, and the solenoid cannot hold in a pulled position.  After cutting the spring in half the solenoid can pull and hold with manual assistance.  But the detent doesn't rise to the up position.

### Screenshots/Images/Files

[image]

[image]

### Applicable Part Numbers

OH4A9-1

### Release Version

v0.2.0

### Category

Mechanical (Structure/Panels/Mechanisms), Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-03-20:
We should only need about .1875 travel. Can I request you cut some loops off the spring incrementally and see if you can get it to a length that it still reliably extends the detent into the path of the stops, and reduces the force significantly?

Arribe, 2024-03-20:
Cut the spring with about a 60/40 split.  neither are long enough to hold the detents high enough with the Solenoids mounted all the way up.  I did try stretching it out a bit to see if that would work to hold the detents high enough up (didn't really work)...but still too much strength on the spring, and solenoids are a bit low.