jrsteensen/OpenHornet #1044: [Bug]: 4A9A1-THROTTLE_CONTROLLER Solenoids Always On

Arribe, 2024-04-02:
### Discord Username

Arribe

### Bug Summary

When external power is applied to the Throttle Controller PCB, the solenoids are powered on.

### Expected Results

The solenoids should be able to turn off when the signal pin is set LOW.

### Actual Results

The solenoids are always on.  I tried the very simple code below, and removing the ESP32 from the controller socket.  The solenoids remain on.  They also remain on when power cycling the PCB.

#define MAX_SOL_SIG 39   ///< MAX_SOL_SIG
#define IDLE_SOL_SIG 37  ///< IDLE_SOL_SIG

void setup() {
  // put your setup code here, to run once:
  pinMode(IDLE_SOL_SIG, OUTPUT);
  pinMode(MAX_SOL_SIG, OUTPUT);

  digitalWrite(IDLE_SOL_SIG, LOW);
  digitalWrite(MAX_SOL_SIG, LOW);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(IDLE_SOL_SIG, LOW);
  digitalWrite(MAX_SOL_SIG, LOW);
}

### Applicable Part Numbers

4A9A1-THROTTLE_CONTROLLER

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Throttle

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-04-08:
Evaluate PCB circuit to see if there is a PCB error.

kbastronomics, 2024-04-13:
I think you can resolve this with a different MOSFET.
VGS is what turn's on/off the MOSFET,  in this case you're using a P-CHANNEL Mosfet

"Vgs is the voltage applied between the gate and source terminals of the MOSFET. In MOSFET operation, Vgs is a critical parameter that determines whether the MOSFET is in the "on" or "off" state. For an NMOS MOSFET, a positive Vgs turns the device on, while for a PMOS MOSFET, a negative Vgs is required."

so PMOS (P CHANNEL) needs a VGS that is Positive with respect to the Drain.   the Drain here is +5V and the Gate is tied to Ground.   (since the gate is negative with respect to the drain it's on.   the ESP32 is a 3.3V IC,   3.3V is still negative with respect to the Drain.

[image]

you need a -0.55v (4.45v with a drain of 5v ) to turn it on which the ESP32 cannot deliver.

you can see here in a schematic used for this IC

[image]

they tie gate high not low

[image]

simple solution is find a MOSFET with a VGS that allows a 3.3v level to work and replace the MOSFET with it.

**You should be able to test this Appling 5V (12V would be better to be sure we've meet the conditions for VGS) to the gate. it should then turn off** 

the PCB fix would be to do the same if you did not want to change the MOSFET

changing the MOSFET is an alternative.