jrsteensen/OpenHornet #1056: [FEATURE REQUEST]: PCB for Radar Altimeter does not match Neopixels as well as Red/Green LED should have some PCB

kbastronomics, 2024-05-07:
### Discord Username

Sandra

### Feature/Enhancement Summary

The Radar Altimeter is very difficult to wire in its current state as you have to run 8 short jumpers from the 2 5mm Neopixels to the pcb.   Also the Red/Green LED are same requiring running wires in a very tight space.
a redesign of the Backlighting to allow for easier installation as well as the warning led's would make the design much more user friendly.

### Screenshots/Images/Files

[image]

[image]

### Justification

Wiring these area's is difficult due to space limitations making it difficult to successfully implement these LEDS

### Applicable Part Numbers

OH5A2A4-1

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Right Console (RCON)

Ben-F111, 2024-05-12:
I have as discussed prior, we have re made this PCB to move the NEO pixel conection to match the Bezel - note we use a standard LED, not NEO PXL now for the OZ Horned, but placment is the same location 

![image](https://github.com/jrsteensen/OpenHornet/assets/67782885/7ecf
[GAUGE_Analog Instruments-12v.zip](https://github.com/jrsteensen/OpenHornet/files/15287696/GAUGE_Analog.Instruments-12v.zip)
7849-b6fa-47e3-8830-aa56d5a65145)

kbastronomics, 2024-05-12:
> I have as discussed prior, we have re made this PCB to move the NEO pixel conection to match the Bezel - note we use a standard LED, not NEO PXL now for the OZ Horned, but placment is the same [...]

Was adding the arc at the top of the LED ARM (lack of a better term) work for making it fit in the case.   
when I looked it appears like there's not enough room actually within the case and so it would not fit.

kbastronomics, 2024-05-12:
[image]

Instead of a new PCB.  could we change the 3D Parts and bring the LEDs into alignment with the current PCB?
Pretty sure the answer is no, as it looks like they would colide with other parts

Ben-F111, 2024-05-12:
did not note a clacsh in the physical build, can take more pics later if required, note the PCB I use has allowed for the screw below 

[image]

Ben-F111, 2024-05-12:
[image]