jrsteensen/OpenHornet #1060: [Bug]: Standby analog instrument PCBs: photosensor connections

Higgins-OH, 2024-05-12:
### Discord Username

Higgins

### Bug Summary

The PCBs of VVI/Airspeed and Standby Altimeter all have 1x3 pin headers (according to Fusion plans and BOM/CPL) on the face side to connect the photoswitch. The combined length of the Dupont connector and the pin headers exceeds the available room between PCB and dial face.

The util_photoswitches also have a 1x3 pin header (not in Fusion, but in the BOM/CPL) which, when used with a Dupont connector, blocks the zero wheel.

### Expected Results

The pin headers of the PCBs can be put on the back of the PCB. The cables can be soldered directly to the photoswitch. This way there is still a pluggable connection between the two, removing the mentioned problems.

### Actual Results

Zero wheels blocked due to photoswitch Dupont connector.
Dial faces not mountable due to PCB Dupont connector.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OHC0007-14 et al.

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Ben-F111, 2024-05-12:
do not install pins, these are to be direct solder of the wires 

[image]

jrsteensen, 2024-07-26:
Add drawing note somewhere to not install connector to photoswitch PCB (as was original design intent.)

jrsteensen, 2024-07-26:
Closed with #1076

closed 2024-07-26 by jrsteensen