jrsteensen/OpenHornet #1061: COMMs C7D0123N SWITCH, ROTARY, 15 DEG, 23 POS, - Config

Ben-F111, 2024-05-12:
### Discord Username

Ben-F111

### Bug Summary

During ASSEM of the OZ Hornet Cockpit COMMS Panels, it was noted that the COMMS ILS Knobs are not confirmed to be Break before MAKE, thus there is a possibilty to breifly short 2 adjacent inputs while turning 

even though this switch is sold as NON-Shorting, it is not Guaranteed !

NOTE the WARNING on the data sheet.

SHORTING AND
NON-SHORTING
SWITCHING IS
NOT GUARANTEED
ON THE 15° INDEXING
C7 SERIES.
https://www.electro-nc.com/sites/electro-nc/files/c7.pdf

[image]

### Expected Results

 there is an ability to short 2 adjacent inputs while turning 

### Actual Results

by changing the matrix to be all ODDS SW numbers on Array 1 and all EVEN SW numbers on Array 2, it is no longer possible to short pins on the same array that are next to each other

### Screenshots/Images/Files

Nil

### Applicable Part Numbers

OH4A7A1-1 and OH-INTERCONNECT (LEFT)

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

Nil

jrsteensen, 2024-07-26:
Implement this change, and create an issue in jrsteensen/OpenHornet-Software to update sketch referring to OH4A7A1-1 with new pinout.

jrsteensen, 2024-07-27:
@Ben-F111 So after looking at the datasheet, I'm already saying to do this - Per the switch datasheet, pin 1 is deck 1, pin 2 is deck 2, pin 3 is next location CW on deck 1, so on. Do we need a change? Or just a note clarifying on the interconnect (even though people should be looking at datasheets?)

Ben-F111, 2024-07-28:
> @Ben-F111 So after looking at the datasheet, I'm already saying to do this - Per the switch datasheet, pin 1 is deck 1, pin 2 is deck 2, pin 3 is next location CW on deck 1, so on. Do we need a [...]

if you fell covered, happy to close it, yes, just a note from our build

jrsteensen, 2024-09-09:
Already covered by documentation. Closing this one.

closed 2024-09-09 by jrsteensen