jrsteensen/OpenHornet #1062: [Bug]: Different Electrical Requirements on OH homepage (Dev Wiki) and interconnect PDF

Higgins-OH, 2024-05-19:
### Discord Username

Higgins

### Bug Summary

The Dev Wiki on the OH homepage isn't up-to-date.

### Expected Results

Same information on Wiki and interconnect.

### Actual Results

See pictures.

### Screenshots/Images/Files

[image]

[image]

### Applicable Part Numbers

Dev Wiki, Interconnect v12

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Top Level Assembly (TLA)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2024-05-19:
26awg is much better for signal cables (not counting those in the BACKLIGHT and/or the BUSS cables.
but wiring from the ALE/ALE Relays to the panels 26AWG should be enough

jrsteensen, 2024-07-26:
Fixed

[image]

closed 2024-07-26 by jrsteensen