jrsteensen/OpenHornet #1083: [Bug]: multiposition rotary switches: Every position needs a connection.

Higgins-OH, 2024-08-07:
### Discord Username

Higgins

### Bug Summary

Since it isn't possible (at least right now) to program the multiposition switches for DCSBios in a way that the "off" position is recognized without a connection, every single position needs to be connected. The switches missing one connection with available pins on the board are as follows:
- ECM (additional pin for "off" available, for instance by connecting ECM_JETT_SW to D4-A6_J3-6 and optional GND to GND_J3-10)
- RWR Control (additional pin for "RWR_DIS_N" available with D16_J4)
- BLEED AIR (additional pin for "NORM" available if ENG_ON and ENG_TEST are moved to J3)

Connection missing, but not enough available pins:
- RADAR & INS (only one available pin on board: D2; two are needed)

### Expected Results

Every position needs to be recognized by DCSBios. Connection needed.

### Actual Results

The "off" position isn't connected.

### Applicable Part Numbers

OH-Interconnect (here: v13)

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Top Level Assembly (TLA)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Arribe, 2024-08-07:
Issue is in the Software repository, and logged with skunkworks. DCSBios has a PIN_NC option allowing for a single switch pin to not be connected. DCSBios doesn’t handle it correctly, so we’ve coded around that in the current software sketches that have these switches.

closed 2024-08-07 by jrsteensen

Higgins-OH, 2024-08-07:
Alright, I'll give it a try. If it works flawlessly, then the comment can be removed (5A5A1-ECS_PANEL.ino, develop branch):

"@bug The Bleed Air knob rotation bounces to NORM when rotated, potentially needs a hardware resolution to connect the NORM position to an Arduino pin.
*/"

Arribe, 2024-08-07:
Not 100% sure about removing the comment. Out of all of the rotary switches the bleed air knob is the toughest for me to rotate within the debounce window.  The knob doesn’t have a definite click and from months of use I’ve had it bounce around. In practice it’s not that big a deal, because having it bounce between positions while rotating doesn’t really mess up the sim like if the INS bounces to off and the Jett losses qual. If others have a different experience for bleed air then we can remove the bug, until then I was just calling it out for awareness.

kbastronomics, 2024-08-09:
> Issue is in the Software repository, and logged with skunkworks. DCSBios has a PIN_NC option allowing for a single switch pin to not be connected. DCSBios doesn’t handle it correctly, so we’ve coded around that in the current software sketches that have these switches.

What was the issue that was opened with DCS-BIOS,  do they plan a fix?
looking at the "FIX" here, any change to DCS-BIOS could break it including a change to PIN_NC.

kbastronomics, 2024-08-09:
it was #57, and #58 they've not accepted or done anything with either