jrsteensen/OpenHornet #1084: [Bug]: 3D models for switch TL1240GQ on the DDI pcb are in the wrong orientation in KiCad

MartelSimon, 2024-08-11:
### Discord Username

1_neox_1

### Bug Summary

The 3D models for the switch TL1240GQ inside KiCad for the DDI pcb are in the wrong direction and are misleading.

### Expected Results

The 3D models for the switch TL1240GQ inside KiCad for the DDI pcb should follow the manufacturer datasheet. (https://www.mouser.ca/datasheet/2/140/TL1240GQ-3458821.pdf https://eswitch.wpengine.com/wp-content/uploads/2022/06/TL1240.pdf)

### Actual Results

The long lead (anode according to the manufacturer datasheet) on the 3D model is not on the positive source and is on the cathode position / GND.

### Screenshots/Images/Files

[image]
[image]
[image]
[image]

### Applicable Part Numbers

OH1A3-1

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [ ] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2024-08-11:
Verified the 3D model is rotated 180 degree's from where it should be.
i already have the PCB file updated and will submit a pull request for it.

jrsteensen, 2024-09-09:
@kbastronomics If you could go ahead and make the changes to MCAD by integrating the new PCB in the model. If any drawings are Rev - that are impacted, please rev them to Rev A as well, then submit the updated ECAD & release package drawings. No Master BOM updates should be required.

jrsteensen, 2026-04-08:
Merging the linked PR, but still need F360 updates.