jrsteensen/OpenHornet #1092: [FEATURE REQUEST]: Radar Altimeter needs more screws for bezel/stepper mount stability

Higgins-OH, 2024-09-07:
### Discord Username

Higgins

### Feature/Enhancement Summary

The complete Radar Alt assembly is held by two screws only. 
To make things worse, these screws need to be installed quite loosely to reduce friction on the indexer ring.
If the knob is pushed (to test), it becomes obvious that a third screw is needed, because only gravity keeps the body in place. Preferrably at the bottom of the stepper mount.

### Screenshots/Images/Files

1: The only screws to connect bezel and body.
[image]

2: Effect of pushing knob when screws are loose enough for the indexer ring to move.

https://github.com/user-attachments/assets/0154e454-627a-4b8f-9f9c-8fbe6a5736db

### Justification

The whole assembly will be more rigid and at the same time allows the indexer ring to move.

### Applicable Part Numbers

OH5A2A4-1

### Release Version

v0.2.0

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

kbastronomics, 2024-09-07:
i did a quick look and i do not see anywhere an additional mounting hole could be added and have enough integrity to be stable.  I tried a couple of things in fusion to add space but anytime I make the mounting hole the correct size it blows away a edge or something.

be interesting if anyone else can come up with a method

Higgins-OH, 2024-09-07:
What if the stepper mount were a bit extended in one corner? It's similar to the upper screw and would definitely help.
The Off Flag wouldn't be affected.
[image]

kbastronomics, 2024-09-07:
yes, that's what I tried,  there is only one spot I can see where a hole can be put that goes thru to the bezel and that spot it gets really thin on the bezel once a insert is put in place