jrsteensen/OpenHornet #1094: [Bug]: OHC0007-14 Misnamed Folder in Release Package

Arribe, 2024-09-07:
### Discord Username

Arribe

### Bug Summary

The PCB manufacturing folder name is listed as...release/OHE_Electronics/OHC0007-14 - PCB, STANDBY AIRSPEED_VVI.  The part name in the OHC0007-1 pdf is "2 IN GAUGE STEPPER PCB".  The PCB is used in the standby cabin pressure gauge in addition to the Standby Airspeed indicator and VVI.

### Expected Results

The folder name should match the drawing's description for OHC0007-14: PCB, 2 IN GAUGE STEPPER

### Actual Results

Folder is named OHC0007-14 - PCB, STANDBY AIRSPEED_VVI

### Screenshots/Images/Files

[image]

[image]

### Applicable Part Numbers

 OHC0007-14

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Top Level Assembly (TLA)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-09-09:
@Arribe Is this also referred to in any other drawings where this PCB is used as the correct description?

Arribe, 2024-09-09:
Don’t think the drawings are wrong, since the should be linked to the part. It’s the folder name that someone manually created.

closed 2024-09-09 by jrsteensen