jrsteensen/OpenHornet #1095: [Bug]: HUD PCB Pot Alignment holes incorrectly placed

Krikeee, 2024-09-08:
### Discord Username

Krikeee

### Bug Summary

The alignment holes for the POTs on the HUD Panel PCB are incorrectly placed by ~180 degrees. This causes the knobs to be misaligned when they are attached.

### Expected Results

Knobs should line up with the legend plate correctly. it is important to note the D shape of both the shaft and the knobs to be sure the change will correct the issue.

### Actual Results

Currently the pots need to be rotated ~180 degrees to correctly align the knobs with the legend plate. 

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH1A7-101

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

This change may require relocating the LEDs on the PCB to make room for the new hole location (this is just a casual observation and no measuring was done). A viable work around (which I used) is to remove the alignment pin on the POT and rotate it as needed using a lock washer to secure it in place.

kbastronomics, 2024-09-09:
Looking at the PCB layout it may be possible to fix this but it's not just a PCB rotating the antirotating hole but moving WS2812s,  also changes to the backplate are needed,  I've only looked at the AoA so far,  all 4 need adjustment.

for now the recommendation is the clip the anti rotation tab on the pot and turn it to the correct orientation

kbastronomics, 2024-09-09:
P0915N-FD20BR10K would also resolve the issue but can not find online

jrsteensen, 2024-09-09:
Image by Sandra showing the pots clocked on the current PCB looking at LED interference:
[image]

Given that we are trying to standardize the pots as much as possible, a PCB/Backplate update is probably the correct path forward marching towards a full release status. Moving the LEDs a few mils up on the PCB probably won't be detrimental.

The 180° clocked pot, or clipping the rings is a sufficient workaround for those with this version.

MartelSimon, 2024-09-09:
There is more than just the HUD panel with this issue, I know the UFC also has it. I will try in my free time to make a bug report for each of them. I can confirm that filing off the alignment tab does work perfectly and is easy to align them without it.

closed 2026-09-17 by jrsteensen