jrsteensen/OpenHornet #1136: [FEATURE REQUEST]: LIP More holes in subfloor brace(s) for cable management

Higgins-OH, 2025-02-22:
### Discord Username

Higgins

### Feature/Enhancement Summary

There are a lot of cables that have to lead from the left side of the LIP to the right LIP bulkhead. 

Either they have to cross the upper area of the foot space (which is already cramped by many cables) or laid on the floor, in which case they interfere with the linear actuator of the rudder pedals.

If there were more holes in the subfloor brace, the cables could be hidden from view and would not be in the way of the rudder pedal adjustment system. There is a small gap between linear actuator and brace, but it is hardly wide enough for the bulkhead cabling.

### Screenshots/Images/Files

[image]

### Justification

A better way to hide cables to the right fwd bulkhead.

### Applicable Part Numbers

OH2A1A1-19 subfloor brace

### Release Version

v0.2.0

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

eHanseJoerg, 2025-02-26:
I second that. In addition to Higgin's sketch, I would consider to increase the whole sizes on marked locations below. 

[image]

kbastronomics, 2025-02-26:
instead of changing the structure why not add a HUB on the right side so you have 1 cable for USB going there.

I would not increase the hole side if that route was taken,  I'd add more holes instead of

Higgins-OH, 2025-02-26:
The cables leading from left to right are ABSIS BUS (6), BL (2x4), USB and additional cables in the future. 
There is no need for an additional USB hub. And, since every hub reduced the maximum number of USB devices on one USB bus, I would strongly advice against it.

kbastronomics, 2025-02-27:
well,  I would not daisy chain hubs period personally, and I was not meaning to daisy chain another hub,   add another direct hub to the PC for the right console.  add a second USB in on the left console plate,  pretty sure there's room, for it and route that one to the right console.   better would be to add a cutout on the right side of the LIP and just replicate the left side input panel.

Higgins-OH, 2025-03-12:
After having drilled these holes I would like to redraw the request:
Handling cables through the holes with a fully assembled LIP is a nuisance. The sharp angles of the cables are not great on wear and tear. In my case there was enough space between linear actuator and wooden plate to contain the cables - but since I wasn't able to build complete by spec, more input by other builders is needed.

eHanseJoerg, 2025-03-14:
[image]

While we are at it. Further holes at the first strut would enable builders to add latches. The picture above shows my "tactical revision" after building to OH spec first and then later discovering that I need to connect the modules in a sturdy way, given the rudder forces and body movements.