jrsteensen/OpenHornet #1154: [Bug]: Electrical backfeeding on backlight module

eHanseJoerg, 2025-06-12:
### Discord Username

Ulukaii

### Bug Summary

With only USB connected, I observe electical backfeed:

- 12V LED on the BLM still lights up
- 12V LED on the ATX PSU breakout board still lights up
- Arduino Mega downstream on ABSIS Bus is powered
- Arduino Pro Micro downstream of the Mega on ABSIS Bus is powered.

I then tried to understand if the root cause of the problem is within the MEGA or within the board. My conclusion is: both. Here is a summary of my findings (thanks Claude):

Observations:
 

- When only USB is connected to the Mega 2560, the 12V LED on the breakout board lights up
- The board is backfeeding 3V-4V through the 12V line
- This occurs even when no external 12V supply is connected

Identified Root Cause:
 

- The Arduino Mega 2560's voltage regulator (NCP1117) remains active when powered via USB
- The 5V from USB is being regulated and outputting a voltage on the 12V line
- This creates an unintended power path through what should be an unpowered 12V line

Problems:

- Components that should only be powered by actual 12V are receiving 3V-4V
- This can cause:
- Unpredictable behavior in components
- Potential damage to components
- **Safety hazards**
- Incorrect state detection
- Components being partially powered when they should be off

Possible Solutions:
 

1. Schottky Diode: Add a Schottky diode in series with the 12V line. Benefits:
 -  Low forward voltage drop (typically 0.3-0.5V)
 - Fast switching
 - Good current handling
2. Power MOSFET Solution: use a MOSFET as a switch, only enabling power when real 12V is present.
3. Voltage Supervisor Circuit: Implement a circuit that only enables the 12V line when proper 12V supply is detected.

### Expected Results

I would have expected that the BLM PCB protects the circuits from backfeeding.

### Actual Results

Connecting a USB will cause backfeeding of 3V-4V on the 12V line

### Applicable Part Numbers

ABSIS_Backlight Controller

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [x] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2025-08-16:
my checking shows this issue originates from the Mega2560 board itself and not the backlight control board.
a check of the published schematic does not match the actual PCB,,   by the schematic there should be no backfeed possible however  when pluging into USB I can see 5V being put onto the RAW IN pin so somewhere there is something connecting the vUSB to the RAW ping

Ultramarine00, 2025-08-16:
Just commenting here to say that I am also experiencing the same back feeding issue as outlined above.