jrsteensen/OpenHornet #1158: [Bug]: OH5A2A2 - Incorrect assembly instructions on Wing Fold Lever drawing

paradoxx42, 2025-07-16:
### Discord Username

paradoxx_42

### Bug Summary

As of the current drawing, the detents do not face the spring plungers. They need to be rotated by 90° clockwise. The second collar with the set screw #21 needs to be rotated by 90° counterclockwise for the handle to lock at the extended wing position.

### Expected Results

The assembled lever should function as in the simulator, it turns to the right and can be pushed at that position.

### Actual Results

The shaft can not fit into the body as in the drawing and has to be rotated by 90° clockwise to interface with the spring plungers. This makes the handle be incorrectly turned by 90°.

### Screenshots/Images/Files

[image]
as designed:
[image]
the outcome:
rightmost locked position
[image]
leftmost position
[image]

### Applicable Part Numbers

OH5A2A2-(12,18)

### Release Version

v0.2.0

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [ ] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [x] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

- redesigned the endcaps (OH5A2A2-15,17) to fit the shaft (OH5A2A2-13) directly without any bushings.
- printed the collars instead of using steel and added a heat set insert for the keyway
- changed some mounting hole diameters
this should not change the issue I was addressing

paradoxx42, 2025-07-16:
What I forgot to mention is that the designator #22 for both collars is missing.

eHanseJoerg, 2025-07-26:
I can confirm the issue. Just built the wing fold handle and it moves incorrectly if assembled according to the drawing.