jrsteensen/OpenHornet #1166: [Bug]: Incorrect connector specified for ABSIS05-10

Benny927, 2025-08-16:
### Discord Username

Murtle

### Bug Summary

The OH interconnect specified connector to the backlight 5 pos conn header (j14) is SMP-05V-BC. This is the wrong connector and should be PN. 0022013057 CONN RCPT HSG 5POS 2.54MM.

### Expected Results

Interconnect to include correct connectors.

### Actual Results

Interconnect drawing has an incorrect features.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

ABSIS005-10 - PCB, ABSIS_BACKLIGHT CONTROLLER

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [x] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2025-10-05:
- [ ] Update interconnect
- [ ] Update interconnect BOM

jrsteensen, 2026-09-19:
- [ ] Specified connector should be 50-29-1900 (CONN, RCPT, HSG, 5POS, 2.54MM, MOLEX KK 254) not 0022013057
  * Connector on backlight controller should be 22-27-2051 not standard pin header.

closed 2026-09-19 by jrsteensen