jrsteensen/OpenHornet #1167: [Bug]: Inconsistancies in the PSU requirements

breithzell, 2025-08-16:
### Discord Username

Breith#3726

### Bug Summary

There are inconsistencies in the minimum PSU requirements between several files

### Expected Results

The minimum values and the suggested PSU in the Fusion model should be consistent.

### Actual Results

- Fusion File (LIP): CP-9020195-NA - RM Series RM750 (750W)
- ABSIS_ATX Power Supply Unit: "Note: Quality ATX PSU of 650W or greater should be used"
- OH-INTERCONNECT - OPENHORNET SYSTEM INTERCONNECT_A_v15: "2A14A2 ATX PSU (800W MIN)"

### Screenshots/Images/Files

- Fusion File (LIP): 

[image]

- ABSIS_ATX Power Supply Unit: 

[image]

- OH-INTERCONNECT - OPENHORNET SYSTEM INTERCONNECT_A_v15: "2A14A2 ATX PSU (800W MIN)"

[image]

### Applicable Part Numbers

2A14A2

### Release Version

v0.2.0

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [ ] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2025-10-05:
I agree the information between the MCAD and interconnect conflict, but I do not think the schematic for the ATX PCB conflicts since that may be used outside of OpenHornet.

[ ] Update note on interconnect to say 750W MIN

breithzell, 2025-10-07:
I'm a bit confused by "that may be used outside of OpenHornet". Yes, sure, in the end it's basically only a power distribution PCB, but that file is part of the ECAD folder, and more specifically the ABSIS folder and it's made in order to have as many headers that what the OpenHornet project needs. 

Even if it can be used for something else, this is the construction files for a component of the ABSIS bus for the OpenHornet. If the requirements for OpenHornet is 750W this should show the minimum required for this project, or specify that the card itself works with 650W, but we need more to run the ABSIS bus.